linux-nvme/nvme-stas #3: D-Bus configuration files placement

igaw, 2022-01-27:
Usually distribution place the D-Bus configuration under `${datadir}/dbus-1/system.d`
nvme-stas seems to hard codes the placement to `/etc` 

  `dbus_conf_dir = join_paths(etcdir, 'dbus-1', 'system.d')`

Would it be possible to make this more packaging friendly?

martin-belanger, 2022-01-27:
Absolutely. This is the first time I'm creating a package intended for distros. I welcome any suggestion like this one.

igaw, 2022-01-27:
Don't assume that I have any ideas what I am doing. It's just one thing I had to patch. The other thing I had to touch is the interpreter path: 
https://build.opensuse.org/package/view_file/home:wagi/nvme-stas/nvme-stas.spec?expand=1

martin-belanger, 2022-01-27:
Resolved by https://github.com/linux-nvme/nvme-stas/pull/5

closed 2022-01-27 by martin-belanger