linux-nvme/nvme-stas #4: D-Bus configuration files allow receive_*

igaw, 2022-01-27:
The package lint check says

```
[    4s] nvme-stas.x86_64: W: dbus-policy-allow-receive <allow receive_sender="org.nvmexpress.stac"/> /usr/share/dbus-1/system.d/org.nvmexpress.stac.conf
[    4s] nvme-stas.x86_64: W: dbus-policy-allow-receive <allow receive_sender="org.nvmexpress.stac"/> /usr/share/dbus-1/system.d/org.nvmexpress.stac.conf
[    4s] nvme-stas.x86_64: W: dbus-policy-allow-receive <allow receive_sender="org.nvmexpress.staf"/> /usr/share/dbus-1/system.d/org.nvmexpress.staf.conf
[    4s] nvme-stas.x86_64: W: dbus-policy-allow-receive <allow receive_sender="org.nvmexpress.staf"/> /usr/share/dbus-1/system.d/org.nvmexpress.staf.conf
[    4s] allow receive_* is normally not needed as that is the default.
```

martin-belanger, 2022-01-27:
What tool do you use to lint with? Thanks.

igaw, 2022-01-27:
No idea. This is from our openBuild service:

  https://build.opensuse.org/package/live_build_log/home:wagi/nvme-stas/openSUSE_Factory/x86_64

martin-belanger, 2022-01-27:
Resolved by https://github.com/linux-nvme/nvme-stas/pull/5

closed 2022-01-27 by martin-belanger