linux-nvme/nvme-stas #40: D-Bus interfaces suggestions

tbzatek, 2022-03-11:
I finally managed to find some time to take a closer look at the `nvme-stas` codebase and would like to point out couple of suggestions from usability point of view:

- [ ] Most of the method calls return a simple string containing some json structure judging by the argument naming. This is quite unfortunate from consumers point of view as they need to take additional steps to parse the JSON structure, e.g. in C apps. Moreover I was unable to find any schema or definition of the returned structure, nor any API stability guarantees. D-Bus offers strong data type system among other objective principles, including complex container types and general Variant type for even more flexibility. Unless the returned JSON structure is free-form and not clearly defined, I would strongly suggest to come up with a concrete return signature.
- [ ] The D-Bus interface XML files should be provided as external files so that binding generators like `gdbus-codegen` can build a convenient ready-to-use client API. Related to the previous point, the benefit of having clearly defined signatures is a direct access to structure members and native data types provided. This can be somewhat done by running `sta[cf]d.py --idl` but it's quite heavy on a build system.
- [ ] And related to that point it's much easier to provide annotations and documentation within the interface XML files...

See also https://dbus.freedesktop.org/doc/dbus-api-design.html and https://dbus.freedesktop.org/doc/dbus-specification.html#container-types

martin-belanger, 2022-03-14:
Hi @tbzatek . Thanks for your comments.

Yes the returned JSON is (was) mostly free-form. It's now starting to "gel" and it is probably time to document this a bit better. I'm aware of Variants, but I find their usage a bit complicated (depending on the language used obviously). 

On the other hand, we're now considering not using D-Bus at all. One of our teams is looking at using `stafd` during early boot (i.e. before the D-Bus broker is running). For that reason, I've been looking into using [Varlink](https://varlink.org/) instead of D-Bus.

tbzatek, 2022-03-15:
> On the other hand, we're now considering not using D-Bus at all. One of our teams is looking at using `stafd` during early boot (i.e. before the D-Bus broker is running). For that reason, I've been looking into using [Varlink](https://varlink.org/) instead of D-Bus.

Oh, interesting. Running any D-Bus daemon within initramfs has always been a source of various problems, being heavy on dependencies aside. I've seen projects switching to different IPC mechanisms for a good reason. Varlink was among them, though not sure how adopted it has become outside the core systemd world.

Varlink brought some ideas onboard from the years of D-Bus experience. Container types, arrays, dictionary, etc. My original suggestion would apply here the same way. E.g. if the output JSON string was supposed to look something like this:
https://github.com/linux-nvme/nvme-stas/blob/62ae55399cf72f0f3ed3d3f79935ef3b058e8162/man/stacctl.xml#L110-L122
with no further nested structures, I would suggest to use return data type of something like `a(a{sv})` in D-Bus signature terms. Corresponding varlink type definition might look even more descriptive and readable on a first sight.

closed 2022-03-24 by martin-belanger

tbzatek, 2022-03-28:
I see, that's what I was afraid of. The structure is way too complex for any reasonable interpretation by standard D-Bus data types. One other way I could think of would be to break it down into actual D-Bus objects and link them using object paths. But that warrants for complete API overhaul and is out of the question for the moment.

Out of curiosity, are there any news regarding the possible switch away from D-Bus due to requirements expressed in https://github.com/linux-nvme/nvme-stas/issues/40#issuecomment-1066755412?