linux-nvme/nvme-stas #46: Dependency on libnvme instead of python3-libnvme?

igaw, 2022-03-23:
Is nvme-stas really depending on libnvme or only only python3-libnvme? I am asking because my packaging build fails to due to 

`meson.build:libnvme_dep = dependency('libnvme', fallback : ['libnvme', 'libnvme_dep'])`

and I don't see any direct dependency on the c library.

martin-belanger, 2022-03-23:
@glimchb - Any thoughts?

glimchb, 2022-03-23:
Sure,

`nvme-stas` is pure python code, so it depends on `python3-libnvme` that depends on `libnvme` 

We didn't have `python3-libnvme` before, so we added `libnvme` dependency

you can change the dependency to `python3-libnvme` in
```
meson.build:libnvme_dep = dependency('libnvme', fallback : ['libnvme', 'libnvme_dep'])
```

Also see spec file here https://download.copr.fedorainfracloud.org/results/@storage/udisks-daily/fedora-rawhide-x86_64/03637253-nvme-stas/nvme-stas.spec
it also looks not very well organized from dependency perspective @tbzatek
```
BuildRequires: libnvme-devel
...
BuildRequires: python3-nvme
...
Requires:      python3-nvme
...
```

martin-belanger, 2022-03-23:
Hi @igaw - Since I cannot reproduce the build failure you're experiencing, could you try this change to see if it fixes the issue?
```
libnvme_dep = dependency('python3-libnvme', fallback : ['libnvme', 'libnvme_dep'])
``` 

And if it does fix the issue, please submit a pull request. Thanks.

glimchb, 2022-03-23:
note python package name changed to `python3-nvme` see https://packages.fedoraproject.org/pkgs/libnvme/

igaw, 2022-03-23:
The `dependency` method used pkg-config and cmake to resolve dependency. This will not work for Python packages.

https://mesonbuild.com/Dependencies.html#dependency-detection-method

Not sure what the general mood in the openSUSE land is about the naming scheme. I don't care at all but I would love to see it align between the distros.

tbzatek, 2022-03-23:
> Also see spec file here https://download.copr.fedorainfracloud.org/results/@storage/udisks-daily/fedora-rawhide-x86_64/03637253-nvme-stas/nvme-stas.spec it also looks not very well organized from dependency perspective @tbzatek

Well `nvme-stas` packaging in Fedora is still work in progress (and `copr` is just an unofficial repo): https://bugzilla.redhat.com/show_bug.cgi?id=2061730
 
> ```
> BuildRequires: libnvme-devel
> ...
> BuildRequires: python3-nvme
> ...
> Requires:      python3-nvme
> ...
> ```

That's simple: `libnvme-devel` rpm provides `libnvme.pc` and it's about the original issue from @igaw. This can go away from my perspective, `python3-nvme` will pull in `libnvme` rpm (for `libnvme.so.1`) automatically. Those `BuildRequires:` are not actually needed, I'll clean that up.

tbzatek, 2022-03-23:
> I'll see if we can talk to someone at RedHat to change the name back to python3-libnvme. There are other projects (although not distributed by distros) that could be confused with python3-nvme (e.g. https://github.com/pynvme/pynvme).

Good point! There's generally quite a bit of confusion when it comes to python module packaging. The original `python3-nvme` naming even got through the package review process. However `python3-libnvme` looks more correct to me, I'll make the change in Fedora while we still can.

igaw, 2022-03-23:
@tbzatek  FWIW, https://build.opensuse.org/package/view_file/openSUSE:Factory/nvme-stas/nvme-stas.spec?expand=1

After looking at your spec file, I think I should also add avahi :)

tbzatek, 2022-03-23:
:-) I generally find SuSE/SLES vs. Fedora/RHEL rpm packaging so different nowadays that it's quite difficult to come with with any universal spec file that actually works...

igaw, 2022-03-24:
Okay, I understood when my libnvme.pc file is correct, pkg-config should work. Let me try to figure out if this works. Thanks for the tip.