linux-nvme/nvme-stas #47: Dependency on python-netifaces

igaw, 2022-03-23:
The upstream project says:

```
Warning
netifaces needs a new maintainer. al45tair is no longer able to maintain it or make new releases due to work commitments.

```
https://github.com/al45tair/netifaces

Are  you really sure you want to depend on this code base?

martin-belanger, 2022-03-23:
I'm using this to verify user mis-configuration. I had cases where our testers would configure stafd to connect to a Discovery Controller by specifying its IPv6 address, but the interface on which they wanted to connect was not configured for IPv6. 

I could remove this dependency, but then I would need to find another way (hopefully as simple as netifaces) to do the same thing. Any recommendations.

igaw, 2022-03-23:
If it is a C program I would pick iproute2 (the helper libraries to be precis) for such things. There seems to be a Python binding for it (https://docs.pyroute2.org/iproute.html) But again SLE15-SP4 doesn't have it packaged.

In the case this code base is not portable, you can use netlink directly and avoid pulling in additional dependencies. https://programtalk.com/python-examples/socket.AF_NETLINK/

martin-belanger, 2022-03-23:
I think I can have the same functionality using psutil. Would `python3-psutil` be a better dependency?

igaw, 2022-03-24:
I think the proper way would be to use AF_NETLINK but it's not simple to get it going without adding a dependency.

Unfortunately, python3-psutil is again not available in SLE15-SP4.

But one thing which might work is just use `ip` directly. `ip` is using AF_NETLINK to get config from the kernel so it's pretty much the best source:

```python
import os                                                                                                       
import json                                                                                                     
import pprint                                                                                                   
                                                                                                                
res = os.popen('ip -j addr').read()                                                                             
data = json.loads(res)                                                                                          
pprint.pprint(data)    
```

```
[{'addr_info': [{'family': 'inet',
                 'label': 'lo',
                 'local': '127.0.0.1',
                 'preferred_life_time': 4294967295,
                 'prefixlen': 8,
                 'scope': 'host',
                 'valid_life_time': 4294967295},
                {'family': 'inet6',
                 'local': '::1',
                 'preferred_life_time': 4294967295,
                 'prefixlen': 128,
                 'scope': 'host',
                 'valid_life_time': 4294967295}],
  'address': '00:00:00:00:00:00',
  'broadcast': '00:00:00:00:00:00',
  'flags': ['LOOPBACK', 'UP', 'LOWER_UP'],
  'group': 'default',
  'ifindex': 1,
  'ifname': 'lo',
  'link_type': 'loopback',
  'mtu': 65536,
  'operstate': 'UNKNOWN',
  'qdisc': 'noqueue',
  'txqlen': 1000},
[....]
```

martin-belanger, 2022-03-24:
Just curious, why did you say that `psutil` is not available in SLES15-SP4.
```
$ uname -a
Linux sles15sp4 5.17.0-rc2-stas-150400.1-default+ #1 SMP PREEMPT Tue Feb 8 06:11:31 PST 2022 x86_64 x86_64 x86_64 GNU/Linux

$ zypper search psutil
Loading repository data...
Reading installed packages...

S | Name           | Summary                                 | Type
--+----------------+-----------------------------------------+-----------
i | psutils        | Tools for Manipulating PostScript Files | package
  | psutils        | Tools for Manipulating PostScript Files | srcpackage
  | python-psutil  | A process utilities module for Python   | srcpackage
i | python3-psutil | A process utilities module for Python   | package
```

igaw, 2022-03-24:
Sure the output of `ip` can change (but it seems pretty stable though), so do libraries change their API. Note the output is in  machine parseable format. This makes it way more reliable to use. But true, it's not perfect. I personally try to avoid adding unnecessary dependency if possible as it also adds up maintenance cost.

About python3-psutil: I tried to look it up via build.suse.com which it doesn't show up. Oh man these distro building is not easy... Well, if python3-psutil is packaged for SLE15-SP4 we should be able to get in without additional problems.

martin-belanger, 2022-03-24:
I decided to remove the code for now. I never liked it in the first place. This came down from our test team who basically want a "psychic system" that can read their minds, determine what they really want to do, and point out the mistake they make when they misconfigure something. Trying to do this is a slippery slope. There is basically no end to such a request. I may instead provide them with a separate tool that can analyze the configuration and try to read their minds... :stuck_out_tongue_winking_eye: 

Issue resolved by: https://github.com/linux-nvme/nvme-stas/pull/49

martin-belanger, 2022-03-24:
@tbzatek - Just FYI. 

Daniel pointed out that `python3-netifaces` was no longer being maintained. So I removed it from nvme-stas.

closed 2022-03-24 by martin-belanger

igaw, 2022-03-24:
Thanks a lot!

Maybe adding something to the how to use/configure documentation might be good enough in this case. 

In a completely different open source project I was contributing, we ended up writing a small shell script which checked all sort of configuration issues and dumped this information. This simplified the help request considerable. Eventually we even coded tips into the script if something was clearly wrong and we were able to detect this from the script. This kept the code base maintainable. Obviously, it's a tradeoff game.

martin-belanger, 2022-03-24:
@tbzatek, @igaw - I just created a new release candidate (rc5). The main change that may impact you is the removal of the dependency on `python3-netifaces`. I also appended `rc5` to the version number in `meson.build` as shown here:
```
 version: '1.0rc5',
```
I wasn't sure if that would impact the creation of RPM packages, so I wanted to give you a heads up.
Regards,
Martin

igaw, 2022-03-25:
I fear it could upset the version comparison logic in rpm. But what do I know. I am absolute a knoob if it comes to packaging. Worst case is that no package update happens until the final 1.0 release happens.

igaw, 2022-03-28:
FWIW, I had no problem with the update. As @tbzatek said, the Version string kept the same format because I was using the tag as input. So all looks good for the SLE15-SP4 integration.