linux-nvme/nvme-stas #76: `make rpm` fails on `version_no_tilde` undefined in the spec file

glimchb, 2022-04-07:
run by
```
git clean -ffdx  && make rpm
```
see
```
...
Created /home/glimcb/Dev/cto/nvme-stas/.build/meson-dist/nvme-stas-1.0.tar.xz
rpmbuild -ba .build/nvme-stas.spec
error: Bad source: .build/meson-dist/nvme-stas-%{version_no_tilde}.tar.gz: No such file or directory
make: *** [Makefile:68: rpm] Error 1
```

martin-belanger, 2022-04-07:
@glimchb - I submitted https://github.com/linux-nvme/nvme-stas/pull/77, which partially fixes this issue. With this change you should be able to get further along running `make rpm`. However, it is still failing as shown below. I can't figure out how to fix it. Hopefully you'll know what to do.

```
rpmbuild -ba .build/nvme-stas.spec
setting SOURCE_DATE_EPOCH=1648080000
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.PYsNZf
+ umask 022
+ cd .build/rpm-pkg/BUILD
+ cd /.build/rpm-pkg/BUILD
/var/tmp/rpm-tmp.PYsNZf: line 32: cd: /.build/rpm-pkg/BUILD: No such file or directory
error: Bad exit status from /var/tmp/rpm-tmp.PYsNZf (%prep)
```

glimchb, 2022-04-07:
@martin-belanger , note `/.build` vs `.build`
try adding `--build-in-place` to `rpmbuild` command

glimchb, 2022-04-07:
or another option is to change/remove
```
%global _topdir       .build/rpm-pkg
%global _sourcedir    .build/meson-dist
```

martin-belanger, 2022-04-07:
Got it to work with https://github.com/linux-nvme/nvme-stas/pull/78

Just be aware that this leaves artifacts in the source tree (i.e. `./redhat-linux-build/`)

closed 2022-04-07 by martin-belanger