linux-nvme/nvme-stas #261: All the connections lost after storage upgrade because the nvme-stas stopped reconnecting after 60 times

igaw, 2022-10-21:
When a storage array needs to reboot due to software upgrade and this takes more than 10 minutes, nvme-stas will drop all connections.

It's a known issue when connecting manually using "nvme connect" and could be avoid by adding "-l -1" to make the retrying infinite.

However, with nvme-stas, it's connected automatically. nvme-stas better to make it infinite as well.

martin-belanger, 2022-10-31:
> When a storage array needs to reboot due to software upgrade and this takes more than 10 minutes, nvme-stas will drop all connections.

It's not that nvme-stas "drops" the connections, but rather that nvme-stas didn't get the "remove" kernel event indicating that the kernel had dropped the connection. 

Testing indicates that the events are generated by the kernel, but nvme-stas is not "seeing" them.

This seems to be happening on systems with lots of NVMe connections (> 150).

martin-belanger, 2022-12-14:
Fixed in nvme-stas 2.0 by https://github.com/linux-nvme/nvme-stas/pull/269

closed 2022-12-14 by martin-belanger