linux-nvme/nvme-stas #300: nvme0 - Registration error. Result:0x0000, Status:0x4002 - Invalid Field in Command

johnmeneghini, 2023-01-04:
Seeing the following errors from stafd when the SFSS controller is reset on the host.

> Jan  4 08:28:43 rhel-storage-09 kernel: nvme nvme0: queue_size 128 > ctrl sqsize 31, clamping down
> Jan  4 08:28:43 rhel-storage-09 kernel: nvme0: Unknown(0x21), Invalid Field in Command (sct 0x0 / sc 0x2) DNR> Jan  4 08:28:43 rhel-storage-09 stafd[1124]: (tcp, 172.18.210.70, 8009, nqn.1988-11.com.dell:SFSS:1:20230103150601e8, 
> enp10s0f0np0) | nvme0 - Registration error. Result:0x0000, Status:0x4002 - Invalid Field in Command: A reserved coded value or 
> an unsupported value in a defined field.

To reproduce:

> 1. generate I/O to all the nvme devices
> 2. Issue resets to all the nvme controllers - example command below:
> 
> echo 1 > /sys/devices/virtual/nvme-fabrics/ctl/nvme0/reset_controller

martin-belanger, 2023-01-04:
Hi @johnmeneghini,
I'm not sure what you mean by "_generate I/O to all the nvme devices_". How do you generate I/O exactly and does it really matter with regards to the reset issue? I also want to clarify what you mean by _Issue resets to "all" the nvme controllers_. Do you mean "all" as in both "Discovery" and "I/O" controllers?

This issue sounds familiar. Didn't we discuss this a while back and concluded that SFSS did not handle the reset command properly (or at all) and then it leaves the connection in a weird state. I'll need to check the status with the SFSS team.

martin-belanger, 2023-01-04:
Hey @johnmeneghini 

Which version of libnvme are you using? There is a bug in libnvme 1.2 that causes a Registration error like the one you described.

johnmeneghini, 2023-01-04:
> Which version of libnvme are you using? There is a bug in libnvme 1.2 that causes a Registration error like the one you 
described.
 >

RHEL 9.2 beta is shipping with libnvme 1.2.

Do you have any idea of exactly what patches are needed from libnvme/nvme-cli?

We can't really update these packages to nvme 1.2.1 at this point, but we might be able to take some patches for a bug fix.

/John

***@***.*** ~]# cat /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="9.2 (Plow)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="9.2"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Red Hat Enterprise Linux 9.2 Beta (Plow)"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:redhat:enterprise_linux:9::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_BUGZILLA_PRODUCT_VERSION=9.2
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.2 Beta"
***@***.*** ~]# yum info libnvme
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

Last metadata expiration check: 3:15:13 ago on Wed 04 Jan 2023 11:25:50 AM EST.
Installed Packages
Name         : libnvme
Version      : 1.2
Release      : 1.el9
Architecture : x86_64
Size         : 219 k
Source       : libnvme-1.2-1.el9.src.rpm
Repository   : @System
 From repo    : beaker-BaseOS
Summary      : Linux-native nvme device management library
URL          : https://github.com/linux-nvme/libnvme
License      : LGPLv2+
Description  : Provides type definitions for NVMe specification structures,
              : enumerations, and bit fields, helper functions to construct,
              : dispatch, and decode commands and payloads, and utilities to connect,
              : scan, and manage nvme devices on a Linux system.

Available Packages
Name         : libnvme
Version      : 1.2
Release      : 1.el9
Architecture : i686
Size         : 89 k
Source       : libnvme-1.2-1.el9.src.rpm
Repository   : beaker-BaseOS
Summary      : Linux-native nvme device management library
URL          : https://github.com/linux-nvme/libnvme
License      : LGPLv2+
Description  : Provides type definitions for NVMe specification structures,
              : enumerations, and bit fields, helper functions to construct,
              : dispatch, and decode commands and payloads, and utilities to connect,
              : scan, and manage nvme devices on a Linux system.

martin-belanger, 2023-01-04:
It's a simple one-liner patch: https://github.com/linux-nvme/nvme-stas/pull/299

johnmeneghini, 2023-01-04:
OK. Thanks Martin.

We'll open a RHEL specific BZ and work on adding this patch.

/John

martin-belanger, 2023-01-09:
Closing. The issue is located in libnvme, where it is fixed already.

closed 2023-01-09 by martin-belanger