linux-nvme/nvme-stas #322: nvme-stas doesn't honor the config JSON file

martin-gpy, 2023-02-23:
Libnvme/nvme-cli can use a config JSON file (schema available at https://github.com/linux-nvme/libnvme/blob/master/doc/config-schema.json) using the -J option for the respective connect-all & connect commands. This is especially useful for large scale systems with several NVMe objects where one needs to apply specific settings to individual subsystems/ports during the respective connect-all/connect in a single go. For e.g.

nvme connect-all -J /etc/nvme/config.json
nvme connect -J /etc/nvme/config.json -n <subsys_nqn> -t <transport> -a <target_IP>

But turns out nvme-stas doesn't handle the same nor provides an option to do so. So looks like this needs to be addressed in nvme-stas itself so that it can process the config JSON file, similar to how libnvme/nvme-cli already does.

martin-belanger, 2023-02-23:
That is correct. nvme-stas does not use the JSON files. Instead nvme-stas provides config files under /etc/stas. This is well documented here: https://nvme-stas.readthedocs.io/en/latest/ (see [stafd.conf](https://nvme-stas.readthedocs.io/en/latest/stafd.conf.html) and [stacd.conf](https://nvme-stas.readthedocs.io/en/latest/stacd.conf.html))

martin-belanger, 2023-02-23:
I initially closed the issue because it claimed that `nvme-stas` does not honor the JSON config that comes with `nvme-cli`. This is true and per design. 

The reasons for that are as follows:
1. `nvme-stas` is independent from `nvme-cli`. One can install `nvme-stas` without installing `nvme-cli`. Making `nvme-stas` depend on a JSON file defined by `nvme-cli` would establish a dependency between the two projects.
2. If `nvme-stas` were to share a JSON file with `nvme-cli`, then we would have a conflict where two services try to perform the same operations in parallel. More precisely, `nvme-cli's` udev rules that automatically apply the JSON config would clash with `nvme-stas` trying to do the same thing.
3. Some users want the ability to configure certain things with `nvme-cli` (e.g. FC interfaces) and configure other things with `nvme-stas` (e.g. TCP interfaces). Therefore, we cannot share a common JSON configuration file between `nvme-stas` and `nvme-cli`.

Please reformulate the issue. For example, instead of saying "_nvme-stas doesn't honor the config JSON file_", maybe you could say "_nvme-stas is missing the ability to set certain configuration parameters defined in nvme-cli's JSON configuration_". Then elaborate exactly what those parameters are and what problem you're trying to solve. 

BTW, I would be more than happy to review a pull request that provides more configuration options, but this PR should not create a dependency between `nvme-stas` and `nvme-cli's`.

martin-gpy, 2023-02-24:
Fair enough.

I'll rephrase the ask here. Basically, `nvme-stas` needs new config options in the /etc/stas/ config files to apply NVMe in-band auth settings (i.e. dhchap_key & dhchap_ctrl_key) while connecting to individual target subsystems.

martin-belanger, 2023-03-03:
FWIW, with regards to in-band authentication there is still a lot of work being done at every layer (kernel, libnvme, nvme-cli). There is also a new spec that just came out, TP8019. There's also work being done for boot. 

I was talking to one of the TP8019 authors and he told me that defining the host key in a JSON file readable by any regular user is not proper. The host key should be protected from prying eyes.

And if we're going to talk about security, we also need to talk about TLS. There's a good chance that the key used for authentication may also be used for TLS handshake. And if that's the case, then that key will probably need to live in a different location (file or other hardware means) than the nvme-cli's JSON config file. 

So, I think it's better to wait and see how things evolve before making significant code changes that may end up being trashed.

igaw, 2023-03-07:
> nvme-stas is independent from nvme-cli

The JSON config parser is part of libnvme. nvme-stas is using libnvme, so I really don't see why you can't use the configs from nvme-cli. If there is something missing, we can add this.

martin-belanger, 2023-03-15:
> > nvme-stas is independent from nvme-cli
> 
> The JSON config parser is part of libnvme. nvme-stas is using libnvme, so I really don't see why you can't use the configs from nvme-cli. If there is something missing, we can add this.

In the JSON config file one can define one or more hosts, each having one or more subsystems, and each subsystem having one or more controllers. 

With TP8009, we can discover Discovery Controllers (DC) using mDNS. The only info we get from mDNS is the DC's IP address, the interface where the DC was discovered, and the DC's NQN (retrieved from the mDNS `TXT` field). Before connecting to the DC one must first create a `root` object and a `host` object. To create the `host` object one must provide the Host NQN and (optionally) the Host ID. If there are multiple Hosts configured in the JSON config file, which Host should we select to be the parent host object for the discovered DCs? 

Currently, I'm assigning all the discovered DCs to the default Host specified by `/etc/nvme/hostnqn` and `/etc/nvme/hostid`. Correct me if I'm wrong, but the JSON config file does not provide a way to define a "default" host, which is needed for the TP8009 use-case.

martin-belanger, 2023-03-15:
@martin-gpy - I've added support for what is currently available in nvme-cli (i.e. Host DHCHAP and Controller DHCHAP). Note however that there is still a lot of work being done with regards to security. New specs were recently made public (e.g. TP8017, TP8019). The keys, which are currently saved to plain text files, will need to be "secured". The kernel keyring may be put to use for that purpose. So the current nvme-cli commands and libnvme APIs are most likely going to change.

All that to say that authentication should be considered "experimental" for the time being.

martin-belanger, 2023-03-16:
Fixed by:
https://github.com/linux-nvme/nvme-stas/pull/335
https://github.com/linux-nvme/nvme-stas/pull/338

closed 2023-03-16 by martin-belanger