linux-nvme/nvme-stas #330: Bad whatis entries for several man pages

bdrung, 2023-03-10:
lintian complains about following:
```
W: nvme-stas: bad-whatis-entry [usr/share/man/man5/org.nvmexpress.stac.5.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man5/org.nvmexpress.stac.debug.5.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man5/org.nvmexpress.staf.5.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man5/org.nvmexpress.staf.debug.5.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man8/stacd.service.8.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man8/stafd.service.8.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man8/stas-config.target.8.gz]
W: nvme-stas: bad-whatis-entry [usr/share/man/man8/stas-config@.service.8.gz]
```

Explanation:

> A manual page should start with a NAME section, which lists the program name and a brief description. The NAME section is used to generate a database that can be queried by commands like apropos and whatis. You are seeing this tag because lexgrog was unable to parse the NAME section.
>
> Manual pages for multiple programs, functions, or files should list each separated by a comma and a space, followed by \- and a common description.
>
> Listed items may not contain any spaces. A manual page for a two-level command such as fs listacl must look like fs_listacl so the list is read correctly.
>
> Please refer to the lexgrog(1) manual page, the groff_man(7) manual page, and the groff_mdoc(7) manual page for details.

martin-belanger, 2023-03-10:
Hi @bdrung - I'm trying to understand what the issue is. I'm not familiar with lintian and the warning message is a bit vague.

Is this because of a bad man page syntax? I'm afraid I'm not familiar enough with the man page syntax to know what the problem is. In fact, that's why I use tools to auto-generate the man pages. If the syntax is wrong, then that must indicate that the tools are broken. By the way, I'm using the same tools that the `systemd` project uses to generate the man pages. I wonder how that project doesn't get issues with lintian.

Would you know exactly what the issue is? Let's take the case of `org.nvmexpress.stac.5.gz` for example. The generated man page is displayed below. 

* I can see that the `NAME` section is present. Maybe lintian is having problems parsing it, but what's wrong with it exactly?
* `Manual pages for multiple programs, functions, or files should list each separated by a comma and a space, followed by - and a common description.` I'm not sure what that means and if that applies to the page below.
* `Listed items may not contain any spaces. `. I don't think there are listed items here (but then again I don't really know what that means.
 
```man
'\" t
.\"     Title: org.nvmexpress.stacorg.nvmexpress.stac
.\"    Author: Mr. Martin Belanger
.\" Generator: DocBook XSL Stylesheets vsnapshot <http://docbook.sf.net/>
.\"      Date: 03/10/2023
.\"    Manual: stafctl
.\"    Source: nvme-stas
.\"  Language: English
.\"
.TH "ORG\&.NVMEXPRESS\&.STACORG\&.NVMEXPRESS\&.STAC" "5" "" "nvme-stas 2.1.4-rc1" ""
.\" -----------------------------------------------------------------
.\" * Define some portability stuff
.\" -----------------------------------------------------------------
.\" ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.\" http://bugs.debian.org/507673
.\" http://lists.gnu.org/archive/html/groff/2009-02/msg00013.html
.\" ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.ie \n(.g .ds Aq \(aq
.el       .ds Aq '
.\" -----------------------------------------------------------------
.\" * set default formatting
.\" -----------------------------------------------------------------
.\" disable hyphenation
.nh
.\" disable justification (adjust text to left margin only)
.ad l
.\" -----------------------------------------------------------------
.\" * MAIN CONTENT STARTS HERE *
.\" -----------------------------------------------------------------
.SH "NAME"
org.nvmexpress.stac
.SH "METHODS"
.sp
.nf
list_controllers (IN  b      detailed,
                  OUT aa{ss} controller_list);
.fi
.SH "DESCRIPTION"
.PP
.PP
.SH "METHOD DETAILS"
.SS "The list_controllers() method"
.sp
.if n \{\
.RS 4
.\}
.nf
list_controllers (IN  b      detailed,
                  OUT aa{ss} controller_list);
.fi
.if n \{\
.RE
.\}
.PP
.PP

.PP
IN b \fIdetailed\fR:
.RS 4
.sp
.RE
.PP
OUT aa{ss} \fIcontroller_list\fR:
.RS 4
.sp
.RE
```

bdrung, 2023-03-10:
Let's look at a good example: The generated man page for nvme-stas.7 has:
```
.SH "NAME"
nvme-stas \- NVMe over Fabrics STorage Appliance Services
```
You have the name "nvme-stas" followed by `-` and the common description "NVMe over Fabrics STorage Appliance Services".

Let's look at an example lintian complains about: org.nvmexpress.stac.5

```
.SH "NAME"
org.nvmexpress.stac
```
It has the name, but no common description. The expected content:
```
.SH "NAME"
org.nvmexpress.stac \- some common description
```

martin-belanger, 2023-03-10:
Thanks for the explanation. Now I just need to figure out how to make the tools generate the right text.

martin-belanger, 2023-03-10:
Should be fixed by: https://github.com/linux-nvme/nvme-stas/pull/336

closed 2023-03-10 by martin-belanger