linux-nvme/nvme-stas #334: How to run test suite against the installed project

bdrung, 2023-03-10:
I am working on packaging nvme-stas for Debian/Ubuntu (see https://bugs.debian.org/1032650). Debian/Ubuntu has autopkgtest for running test against the installed binary package. Can you add documentation how to run the test cases against the installed nvme-stas?

martin-belanger, 2023-03-10:
I'm not familiar with `autopkgtest`. The tests are part of the source repo and are embedded in meson.build files. Are you saying that `autopkgtest` allows one to run tests without having access to the source repo? How does that work? Can you point me to a project where this is used.

bdrung, 2023-03-10:
Some information about autopkgtest:
* https://packaging.ubuntu.com/html/auto-pkg-test.html
* https://salsa.debian.org/ci-team/autopkgtest/blob/master/doc/README.package-tests.rst

You specify which packages you need installed in your test bed and maybe what kind of restrictions you need in your test bed and what to run for the test. When running the test, the current working directory contains the source package content (which is the git export plus the debian directory).

martin-belanger, 2023-03-16:
Hi @bdrung - Hope this will answer your question.

Basic testing is done using `meson`. From the directory where the source package content is located, run the following commands.
```bash
BUILD_DIR=".build"
meson setup ${BUILD_DIR}
meson test -C ${BUILD_DIR}
```
The above commands have been added to the `Makefile` for convenience and can be run simply with:
```
make test
```

Coverage testing is also available, but it is recommended to run that in a VM as this may leave the host computer in a weird state if something were to go wrong. Additional packages are required to run Coverage testing (see: [TESTING.md](https://github.com/linux-nvme/nvme-stas/blob/main/TESTING.md)). The command to run coverage testing is:
```
make coverage
```