linux-nvme/nvme-stas #407: Udev test issues with esoteric network interfaces

ogayot, 2023-11-28:
Hello,

(as discussed in https://github.com/linux-nvme/nvme-stas/pull/406, I'm filing another issue for the sake of completeness)

In Ubuntu, we have an openstack infrastructure that we can use for testing things in VMs (this is different from the infrastructure where autopkgtests run). When running the test-suite in a VM deployed by this openstack, there are multiple udev test-cases failing:

* Test Case 1
* Test Case 8
* Test Case 11
* Test Case 12
* Legacy Test Case D6
* Legacy Test Case F6

What is particular in these machines is the presence of a VXLAN network interface having a link-local IPv6 set (it is also a member of a bridge but I don't think this is relevant). I am not exactly sure what this type of interface does but I managed to reproduce the failure after running the following steps:

```bash
ip link add type vxlan id 1234 dstport 0
ip link set vxlan0 up 
```
I also reproduced the failures with `dummy` interfaces but I'm not sure if that's a real use-case.

```bash
ip link add type dummy
ip link set dummy0 up
```

Thanks,
Olivier

martin-belanger, 2023-11-28:
Fixed by https://github.com/linux-nvme/nvme-stas/pull/409

closed 2023-11-28 by martin-belanger

ogayot, 2023-11-28:
Excellent, thanks for the quick fix!