linux-nvme/nvme-stas #438: Unable to connect to new Discovery Controller on Discovery Controller Change Asynchronous Event Notification

ankkuma, 2025-02-21:
Hi,

I want to connect to a new discovery controller after receiving the AEN for Discovery log change notice.

below are the steps I followed:

1. Make a persistent connection to the discovery controller.

sudo nvme discover --transport tcp --traddr 10.0.0.4 --trsvcid 4420 --persistent

10.0.0.4 => Current NVMF target/ discovery controller.

On Nvmf Host:

[azureuser@NVMe-oF-Initiator ~]$ sudo nvme list-subsys
nvme-subsys1 - NQN=**nqn.2014-08.org.nvmexpress.discovery**
               hostnqn=nqn.2014-08.org.nvmexpress:uuid:142142ac-e66c-41dd-9535-4c6d3466fae4
               iopolicy=numa
\
 +- nvme1 tcp **traddr=10.0.0.4**,trsvcid=4420,**src_addr=10.0.0.5 live**

nvme-subsys0 - NQN=nqn.2014-08.org.nvmexpress:uuid:5a513dee-5df0-4dbb-b063-b4f4b4853c00
               hostnqn=nqn.2014-08.org.nvmexpress:uuid:142142ac-e66c-41dd-9535-4c6d3466fae4
               iopolicy=numa

2. NVMF target sends the AEN using RPC/SPDK.

On target:
sudo ./scripts/rpc.py **nvmf_discovery_add_referral** --subnqn nqn.2014-08.org.nvmexpress.discovery --trtype TCP **--traddr 10.0.0.6 --trsvcid 4420** --adrfam ipv4

AEN detail from target:

async_event_type = SPDK_NVME_ASYNC_EVENT_TYPE_NOTICE;
async_event_info = SPDK_NVME_ASYNC_EVENT_DISCOVERY_LOG_CHANGE;
log_page_identifier = SPDK_NVME_LOG_DISCOVERY;

3. NVMF host received the AEN. It should connect to the new Discovery Controller (10.0.0.6), but it failed to do so.

Feb 21 06:52:25 NVMe-oF-Initiator stafd[355165]: **nvme1 - Received AEN: Change of Discovery Log Page (0x70f002)**
Feb 21 06:52:25 NVMe-oF-Initiator stafd[355165]: Invoking: systemctl start nvmf-connect@--device\x3dnvme1.service
Feb 21 06:52:25 NVMe-oF-Initiator systemd[1]: **Started NVMf auto-connect scan upon nvme discovery controller Events.**
░░ Subject: A start job for unit nvmf-connect@--device\x3dnvme1.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://access.redhat.com/support
░░
░░ A start job for unit nvmf-connect@--device\x3dnvme1.service has finished successfully.
░░
░░ The job identifier is 82439.
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: **ctrl device nvme1 found, ignoring non matching command-line options**
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: Failed to write to /dev/nvme-fabrics: Required key not available
Feb 21 06:52:25 NVMe-oF-Initiator kernel: nvme nvme2: no valid PSK found
Feb 21 06:52:25 NVMe-oF-Initiator kernel: nvme nvme2: no valid PSK found
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: Failed to write to /dev/nvme-fabrics: Required key not available
Feb 21 06:52:25 NVMe-oF-Initiator systemd[1]: nvmf-connect@--device\x3dnvme1.service: Deactivated successfully.

4. **Expectation:** I was expecting host to connect to new discovery controller also which was sent as part of AEN arguments, which we can see that host received it (Verified in the Wireshark traces).

But, it is not happening, host is auto connecting to the same old discovery controller. I was expecting connection to both the discovery controllers.

**What is the solution to this ? Do I need a different AEN ? If we can achieve this with a server side change, that is welcome. Do we need any different Lunix or NVME STAS Version ?**

Version Details:

[azureuser@NVMe-oF-Target spdk]$ cat /etc/redhat-release
Red Hat Enterprise Linux release 9.4 (Plow)

[azureuser@NVMe-oF-Initiator ~]$ rpm -qi nvme-stas
Name        : nvme-stas
Version     : 2.2.1
Release     : 2.el9
Architecture: noarch
Install Date: Sun 26 Jan 2025 03:42:56 PM UTC
Group       : Unspecified
Size        : 623388
License     : ASL 2.0
Signature   : RSA/SHA256, Wed 12 Apr 2023 07:58:31 AM UTC, Key ID 199e2f91fd431d51
Source RPM  : nvme-stas-2.2.1-2.el9.src.rpm
Build Date  : Fri 07 Apr 2023 04:18:36 PM UTC
Build Host  : ppc-061.build.eng.bos.redhat.com
Packager    : Red Hat, Inc. <http://bugzilla.redhat.com/bugzilla>
Vendor      : Red Hat, Inc.
URL         : https://github.com/linux-nvme/nvme-stas
Summary     : NVMe STorage Appliance Services

martin-belanger, 2025-02-21:
Adding @igaw to the discussion since I see that you also reported this issue on the [nvme-cli site](https://github.com/linux-nvme/nvme-cli/issues/2719).

It looks like you are using two different technologies (nvme-cli and nvme-stas) and trying to make them work together. I'm wondering if you could just use nvme-cli instead. This depends on whether you want to configure connections manually (nvme-cli) on all your hosts or automatically with nvme-stas. It also depends on whether you are using a centralized discovery controller (CDC) as per TP-8010, in which case you really want to use nvme-stas. 

You see, nvme-cli (i.e. the `nvme` shell command) is used to configure discovery and I/O controllers manually. Some automation is possible with commands such as "nvme connect-all" and config files (e.g. /etc/nvme/config.json).

On the other hand, nvme-stas was designed to be fully automatic per TP-8009 and TP-8010. The goal is to have a centralized discovery controller (CDC) where a user can control connectivity between Hosts and Storage appliances. In this way, host computers running nvme-stas can be remotely controlled from the CDC so that users don't have to log into each host and use nvme-cli to set up connections. 

If you don't have a CDC, you likely don't need to use nvme-stas. However, nvme-stas was also designed to be used without a CDC. In that case, you need not use nvme-cli but instead you must manually configure your discovery controllers in /etc/stas/stafd.conf.

ankkuma, 2025-02-21:
Hi @martin-belanger,

Thanks for your response. I dont want to configure manually. I want to manage connections automatically using NVME stas.

We are using a CDC, and trying to manage connections on hosts remotely from nvme target. In this case, we are trying to prototype a scenario, where hosts should move/connect to a new discovery controller if the discovery controller itself gets migrated. **Can you suggest how to do that ?** 

Note: It will be a planned migration and NVME target can inform the hosts about the new DC using AENs.

Right now, I was trying to make it work by sending the AEN from nvme target. We were sending "AEN: Change of Discovery Log Page (0x70f002)" from the NVME target to hosts.

martin-belanger, 2025-02-21:
I have a few questions:

1. I just want to be sure that you have a Centralized Discovery Controller (CDC) per TP8010 and not a Direct Discovery Controller (DDC). Is this correct?
2. Does your CDC (or DDC) advertise itself using mDNS (per TP-8009)? 
3. Is the host located on the same subnet as the CDC (or DDC)? 

nvme-stas gets notified by the [avahi daemon](https://github.com/avahi/avahi) when a discovery controller (CDC or DDC) is discovered (or lost). This is what allows stafd to automatically connect to (or disconnect from) discovery controllers (CDC or DDC). Since this relies on mDNS packets, the host and DC must be on the same subnet and the mDNS packets must be allowed to be exchanged (i.e. not blocked by a firewall).

When mDNS is working and the Avahi deamon is able to detect the presence of CDCs or DDCs, then stafd will automatically configure itself and connect to those CDCs or DDCs.

If you don't have mDNS support, then you will need to manually add/remove DCs to/from /etc/stas/stafd.conf. Do not use nvme-cli because that's a different tool. You must use an editor and configure /etc/stas/stafd.conf to add the CDC information. [stafd.conf](https://github.com/linux-nvme/nvme-stas/blob/main/etc/stas/stafd.conf) is self-documented, or you can check the man pages (man stafd.conf), or check the [read-the-docs page](https://nvme-stas.readthedocs.io/en/latest/stafd.conf.html).

ankkuma, 2025-02-21:
Hi @martin-belanger ,

1. Yes,I am using the CDC.
2. I am not sure if I have the mDNS support. I have installed nvme-stas installed on both target and host. avahi-daemon is running on both target and host. Target is running stacd and host is running stafd.

but when I run below, I dont see any output.

Target:
[azureuser@NVMe-oF-Target stas]$ avahi-browse -r _nvme-disc._tcp

Host:
[azureuser@NVMe-oF-Target stas]$ avahi-browse -r _nvme-disc._tcp

**What shall I do to make sure that CDC (or DDC) advertises itself using mDNS (per TP-8009)?**

3. Yes, right now both are located in same subnet. **But, in future we may not have this luxury. In that case, can AENs come to the rescue ?**

Thanks,
Ankit

ankkuma, 2025-02-22:
Thanks @martin-belanger for explanation.

As target already supports sending AEN/RPC to the host and indicates that Discovery Controller migration to new Discovery Controller. Shouldnt Stafd do the rediscover/reconnection to the new discovery controller?

Below AEN specifically adds a referral to new Discovery Controller:

sudo ./scripts/rpc.py nvmf_discovery_add_referral --subnqn nqn.2014-08.org.nvmexpress.discovery --trtype TCP --traddr 10.0.0.6 --trsvcid 4420 --adrfam ipv4

Can this be implemented or fixed ?

igaw, 2025-02-24:
> Can this be implemented or fixed ? This will automate the discovery.

There are/were several attempts to address global fabric discovery for TCP/RDMA (FC has this feature) in the specification. nvme-stas is implementing TP-8009 which relies on mDNS for discovering a CDC. I can't speak for this project, so I don't know if any other means will be added.

Speaking with my nvme-cli upstream hat on:  there are other ways to find a CDC (e.g. it would be possible to do extend the DDC to point to a CDC) but IIRC the spec is not there yet. nvme-cli will only implement feature which are defined in the spec. It's possible to do have some vendor specific solution in the meantime but this is not something I see in nvme-cli.

martin-belanger, 2025-02-24:
If your question is about referrals, then yes, nvms-stas supports referrals. 

If an AEN is received by stafd indicating a change of discovery log pages, stafd will then query the latest discovery log pages. And if there is a new DC in the new discovery log pages (a.k.a. referral to a DC), then stafd will connect to it.

ankkuma, 2025-02-24:
Thanks @martin-belanger .

So, I shared the steps for referral in my first post of this conversation. Pasting a part of that below again for reference. Is there anything wrong or missing in the AEN communication or any other step? I could not get a connection to new DC. Do you want me to share the Wireshark traces of stafd querying and target's response carrying new DC's details?

2. NVMF target sends the AEN using RPC/SPDK.

On target:
sudo ./scripts/rpc.py nvmf_discovery_add_referral --subnqn nqn.2014-08.org.nvmexpress.discovery --trtype TCP --traddr 10.0.0.6 --trsvcid 4420 --adrfam ipv4

AEN detail from target:

async_event_type = SPDK_NVME_ASYNC_EVENT_TYPE_NOTICE;
async_event_info = SPDK_NVME_ASYNC_EVENT_DISCOVERY_LOG_CHANGE;
log_page_identifier = SPDK_NVME_LOG_DISCOVERY;

3. NVMF host received the AEN. It should connect to the new Discovery Controller (10.0.0.6), but it failed to do so.

Feb 21 06:52:25 NVMe-oF-Initiator stafd[355165]: nvme1 - Received AEN: Change of Discovery Log Page (0x70f002)
Feb 21 06:52:25 NVMe-oF-Initiator stafd[355165]: Invoking: systemctl start nvmf-connect@--device\x3dnvme1.service
Feb 21 06:52:25 NVMe-oF-Initiator systemd[1]: Started NVMf auto-connect scan upon nvme discovery controller Events.
░░ Subject: A start job for unit nvmf-connect@--device\x3dnvme1.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://access.redhat.com/support
░░
░░ A start job for unit nvmf-connect@--device\x3dnvme1.service has finished successfully.
░░
░░ The job identifier is 82439.
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: ctrl device nvme1 found, ignoring non matching command-line options
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: Failed to write to /dev/nvme-fabrics: Required key not available
Feb 21 06:52:25 NVMe-oF-Initiator kernel: nvme nvme2: no valid PSK found
Feb 21 06:52:25 NVMe-oF-Initiator kernel: nvme nvme2: no valid PSK found
Feb 21 06:52:25 NVMe-oF-Initiator sh[355422]: Failed to write to /dev/nvme-fabrics: Required key not available
Feb 21 06:52:25 NVMe-oF-Initiator systemd[1]: nvmf-connect@--device\x3dnvme1.service: Deactivated successfully.

martin-belanger, 2025-02-24:
Hi @ankkuma 

Although stafd does see the AEN, it does not take action because the DC that sent this AEN is not configured in stafd's config file. (see **DETAILED EXPLANATION** below).

As I said earlier, you cannot use nvme-cli to configure nvme-stas. This means that in your Step 1, you cannot use `sudo nvme discover --transport tcp --traddr 10.0.0.4 --trsvcid 4420 --persistent`. Instead, edit `/etc/stas/stafd.conf` and add your discovery controller (DC) info under the `[Controllers]` section. And use the proper service ID (`trsvcid`). `trsvcid` should be set to `8009` for discovery controllers and `4420` for I/O controllers.

For example:
```
[Controllers]
controller = transport=tcp;traddr=10.0.0.4;trsvcid=8009
```

**DETAILED EXPLANATION:**
The log shows that stafd is just passing the handling of the AEN to nvme-cli (indicated by this message: `Invoking: systemctl start nvmf-connect@--device\x3dnvme1.service`). 

Note that AENs are sent as udev events by the kernel to userspace applications. The interface from the kernel does not allow stafd to register for a specific AEN from a specific DC. Instead stafd gets AENs from all the DCs, even DCs there were not configured in stafd's config file. In your case, you used nvme-cli to configure the connection to a DC, but this DC is not known to stafd. Therefore. stafd ignores the AEN coming from it. 

stafd only reacts to AENs from DCs that are configured in stafd's config file (or DCs that were discovered by stafd through mDNS).

ankkuma, 2025-02-26:
Thanks @martin-belanger for the detailed explanation. I added the DC host name in the stafd.conf and it started working fine. I am good for now. I will raise a new request in case I have more queries.

Thank you once again for the prompt support.

closed 2025-02-26 by ankkuma